yorikvanhavre/FreeCAD-NativeIFC #2: Object to reference IFC File

carlopav, 2022-10-19:
Just out of curiosity, did you choose to reference the IFC file with an object (instead of using document .Meta) to be able to reference more IFC Files in the same FC document?

https://github.com/yorikvanhavre/FreeCAD-BlenderBIM/blob/feda6a6c9194082359f7c2730a8d1f24025b94f0/bb_import.py#L52

yorikvanhavre, 2022-10-19:
Mostly to mimmick blenderbim, and also to expose all of it as object properties to the user. also, if we take the IFC road, it's good to make the IFC structure faithfully shown in freecad. And finally yes, i think in theory nothing would prevent having more than one project in a file..

yorikvanhavre, 2022-10-19:
also, we might want several ifc files in one freecad file.

Moult, 2022-10-19:
An IFC database should only contain one IfcProject. But nothing is stopping FreeCAD from having multiple IFCs :)

carlopav, 2022-10-19:
So do you imagine this kind of structure in the tree?
```
- IfcDocument
    - IfcProject
        - IfcSite
```

carlopav, 2022-10-19:
It's a bit wide, but i'm perfectly fine with that. +1 to faithfully follow ifc structure. thanks both for the explanation.

Moult, 2022-10-19:
@carlopav note that it is actually optional to show the structure, it is merely a sensible default if the user hasn't provided a choice.

closed 2022-10-20 by carlopav