yorikvanhavre/FreeCAD-NativeIFC #3: Add ifcopenshell to FreeCAD Addon Manager's ALLOWED_PYTHON_PACKAGES.txt

luzpaz, 2022-10-19:
Add ifcoopenshell as a permitted dependency that the addon manager can auto-download.

yorikvanhavre, 2022-10-20:
Is ifcopenshell installable that way? I haven ´t tested...

Moult, 2022-10-20:
If it helps IOS is now available via pip, conda, docker, or manual download: https://blenderbim.org/docs-python/ifcopenshell-python/installation.html

chennes, 2022-10-20:
Thanks @Moult -- added it to the allowed packages list in https://github.com/FreeCAD/FreeCAD-addons/commit/3e83af6b80e6a12dc5fd8ed68a7162a177d9bf35

luzpaz, 2022-10-22:
ifcopenshell is not getting downloaded per user in the forum thread: https://forum.freecadweb.org/viewtopic.php?p=635336#p635336

yorikvanhavre, 2022-11-29:
Tested and everything is now working fine... I think we can close this then!

closed 2022-11-29 by yorikvanhavre