yorikvanhavre/FreeCAD-NativeIFC #5: Get the structure!

yorikvanhavre, 2022-10-20:
This is basically for discussion.

Up to now, this happens:

* Click menu **File -> Open**, select an IFC file, then select the bb_import importer
* The file imports
* One single object is created, which represents one IFC file + project. The shape of the object is calculated from its children

[image]

This is pretty fast, a medium file like schependomlaan (47Mb) imports in 23 seconds on a ryzen9. Further optimizations are possible, like caching, or even using only mesh representation

What should happen next:

* One should be able to right-click the object, and select an **Expand** option
* Upon expanding, the inner site should appear under the project
* The shape of the project becomes instead a compound of its children
* The site can be expanded again, and so on

Other ideas:

* shapes should be cached on initial import, so expand operations just need to make a compound
* alternative import modes so an initial expansion can be done

paullee0, 2022-10-21:
Tested and looks great!

Discussion on FC forum [here](https://forum.freecadweb.org/viewtopic.php?f=23&t=72345&p=634845#p634845)

yorikvanhavre, 2022-10-25:
Done with c2a3b8bd8ddcb34e35a228e6ab4c0516f962606d

closed 2022-10-25 by yorikvanhavre