yorikvanhavre/FreeCAD-NativeIFC #7: roadmap

carlopav, 2022-10-23:
I was hanging around the code this morning and studying a bit BB ifc.py and import_ifc.py.

Shouldn't it be a matter of setting up an IfcStore and substituting Blender objects with FC objects? That is of course an enormous task, but I need to have get the general idea of what we are up to :)

Moult, 2022-10-23:
Here's what I think it might look like:

 1. A singleton to store the active IFC(s?) for the session (i.e. the IfcStore)
 2. Ensure the load process links FreeCAD objects with IFC STEP IDs that are in the active IFC. The mapping should be stored somewhere appropriate for FreeCAD.
 3. Demonstrate how the UI can detect that the user has chosen an IFC-linked object as opposed to a regular FreeCAD object, and is able to use that ID to extract some basic data about that object from the IFC (e.g. basic attributes). This demonstration must show how the UI can cache this data (so that it isn't fetched on every UI draw) and can also refresh the data (e.g. after an operation is done which might change data, it should reload).
 4. Create a simple mockup of how you might create an input form and submit button that is prepopulated with loaded data from step 3. This can be a basic prototype for now but in the future these input forms can be quite clever (e.g. dropdowns, sliders for numeric data types, or even special dedicated UIs for complex data).
 5. Show how submitting the form from step 4 causes the UI to refresh from step 3.
 6. Create a simple "save project" feature, which is basically a file.write() call, as there is no such thing as an "Export" anymore.
 7. Demonstrate how the system knows whether or not a user has edited the geometry of a FreeCAD object. This'll be used later in the geometry syncing strategy.

yorikvanhavre, 2022-10-24:
That's more or less what I have in mind too. Additional comments:

* ATM we have two different kinds of objects: a standard ifc_object, which can represent any IFC object in FreeCAD, and an ifc_document object, which represents an IFC document. I am uncomfortable with the assumption "one FreeCAD document = one IFC document" because it will never be true, users will always be able to add FreeCAD stuff that cannot be exported to IFC. Also, this allows several IFC documents in one FreeCAD document
* both ifc_object and ifc_document are made to display geometry. We could have also non-geometry object types later on
* ATM the ifcopenshell file object is stored in the ifc_document object. I'll look at how the IfcStore works in BB but it should be similar. So basically: One ifc_document contains one ifcopenshell.File, and all other ifc_objects always know which ifc_document they belong to
* ifc_object carries a step id, and, to obtain its geometry, queries its host ifc_document
* there should also be a caching mechanism in place, so geometry doesn't need to be recalculated if nothing changed
* ifc entity attributes are also stored in the ifc_object. Changing any of these attributes automatically triggers FreeCAD functions so we can trace back and change the corresponding value in the host ifc_document

yorikvanhavre, 2022-10-24:
What I get from https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.7.0/src/blenderbim/blenderbim/bim/import_ifc.py :
The IfcStore object basically keeps a cache that is used by the iterator to skip recalculating certain objects, right? It also have some blender-specific functionality and also seems to have its own undo/redo stack, which I suppose is a workaround to integrate with the blender undo system?
@Moult would it be useful in our case? It would need to be stripped from blender-specific stuff. But I was thinking of basically keep a simple shapes cache besides the ifcopenshell.File...

Moult, 2022-10-24:
The ifcstore does not have a cache. It only stores the IFC file object in memory, mappings between blender objects and IFC elements, and a few other small details.

The iterator is clever enough to have its own cache so shared representations are not recalculated. This cache is local to the iterator loop only.

However there is a feature in ifcopenshell to use an external cache file (h5) that can then be reused in future iterators. This is pretty advanced and experimental and I would not recommend looking at it for now.

yorikvanhavre, 2022-10-25:
Okay thanks! So I'll pursue on the current path now (each ifc_document object in FreeCAD keeps its associated ifcopenshell file object in memory), that seems like a stable foundation to build on.

Later on we can refine things and implement some better caching system for the iterator, but that doesn't seems like a blocker for now.

carlopav, 2022-10-26:
The general code architecture is above my understanding skills... But doesnt It looks like we are mailing an good importer module instead of relying on IFC itself as ""

yorikvanhavre, 2022-10-27:
It is exactly that... Each IFC file imported stays linked, and is the generator of all geometry and properties of the document.

When you import an IFC file, an "ifc_document" object is created in the FreeCAD doc. This ifc_document contains an ifcopenshell.File object, which is the opened ifc file. Any object, property or whatever created under it, is directly managed by it. If you change anything to an underlying object, the change is done in the ifcopenshell.File. So you can consider that the FreeCAD objects are actually an interface to the contents of the IFC file itself.

carlopav, 2022-10-27:
ok! thanks for the explanation :)
closing the issue

closed 2022-10-27 by carlopav