yorikvanhavre/FreeCAD-NativeIFC #12: Top level FC object

carlopav, 2022-11-01:
Another question:
Is top level object always an IfcProject?
Does it always have a Type attribute? and if so, its Type attribute is always "IfcProject"?

```
- IfcDocument
    - IfcProject
        - IfcSite
            - IfcBuilding
```
or:

```
- IfcProject
    - IfcSite
       - IfcBuilding
```
Current implementation looks more like the second one:
[image]

yorikvanhavre, 2022-11-01:
Currently there is one FreeCAD object that groups both an IFC file and an IfcProject. I think that makes sense because 1) in one IFC file there is always one and only one IfcProject. So both are actually one thing. And 2) there could be several IFC files in one FreeCAD file, or FreeCAD objects that are not part of IFC. So for me this makes more sense, and I don't see any real advantage of separating the iFC document and the IfcProject. Do you have another opinion? I'd be ourious to hear.. (also @Moult ?)

About its type, at the moment you can change it freely. But maybe in the case of the IfcProject, it should be made read-only... Ore changeable just between IfcProject and IfcProjectLibrary (not sure that's the right name)

carlopav, 2022-11-01:
It's good for me, but I remember last time we discussed it was about having the other solution: https://github.com/yorikvanhavre/FreeCAD-BlenderBIM/issues/2

If the top level object Type is a IfcProject or IfcProjectLibrary i'd like that also the underlying object and viewprovider could be called project or projectlibrary. What do you think?

Moult, 2022-11-01:
Yes in an IFC model there is always one IfcProject, so I see no reason to separate the "document" and "project" (it is also not separated in the BlenderBIM Add-on).

Don't change the type of the IfcProject. If there is one or more IfcProjectLibrary, they exist as a child of the IfcProject.

carlopav, 2022-11-01:
Thx @Moult, so, @yorikvanhavre can we call that object project and not document?

yorikvanhavre, 2022-11-07:
> so, @yorikvanhavre can we call that object project and not document?

Ok for me!!

carlopav, 2022-11-07:
Ok!

closed 2022-11-07 by carlopav