yorikvanhavre/FreeCAD-NativeIFC #15: get properties, materials and other "expandable" structures

yorikvanhavre, 2022-11-10:
This is basically a stub to discuss implementaiton (and for @Moult to explain his visions ;) )

We now have a basic decomposition in the FreeCAD document: Project, site, building, storey, and their direct children. However, there is a lot more to fetch on demand from the IFC file. For ex:

* Properties
* Materials
* Type
* Geometry construction (extrusions, etc...)

In BlenderBIm you basically press some "Edit" buttons to reveal these things in the UI.

I'm not sure how to proceed here. Clearly, materials are FreeCAD objects (stored in a special group), Types too (should be stored in a special group too), geometry constructions should also become FreeCAD objects, and properties and property sets should become object properties (tied to style properties if applicable).

Moult, 2022-11-11:
I think the first thing to try and implement is showing attributes as it's the easiest. Create a panel that is able to lazyload attributes and refresh after operations have been performed.

yorikvanhavre, 2022-11-16:
In FreeCAD we don't have as much control over the properties panel of objects. It can only display object properties. So basically there is no easy way to add something like an "Edit..." button there. What we could do (I'd do that for now, I think) is add a couple of right-click context-menu similar to the current "Expand children", ex: "Expand attributes", "Expand properties", etc.

We could think to something better in the future, but at least the underlying functionality will already be working.

Now, expanding attributes will begin to reveal things like owner history, IfcRel* objects, etc...

How fine / how far should this go? We could in theory show each and every IFC element there, up to the IfcCartesianPoint, with all the inbetweens (IfcRel*, etc...). How did you / do you choose in BlenderBIM?

yorikvanhavre, 2022-11-16:
Side question... How do you get all possible Ifc product types from ifcopenshell?

Moult, 2022-11-29:
```python
schema = ifcopenshell.ifcopenshell_wrapper.schema_by_name('IFC4')
declaration = schema.declaration_by_name('IfcWall')
print(declaration.supertype())
for subtype in declaration.supertype().subtypes():
    print(subtype.name())
```

yorikvanhavre, 2022-11-29:
excellent! thanks!!!

closed 2023-02-08 by yorikvanhavre