yorikvanhavre/FreeCAD-NativeIFC #18: Fails to open IfcOpenHouse

pablo-mayrgundter, 2023-01-13:
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FreeCAD-NativeIFC 2022.10.20
[/code]

[image]

yorikvanhavre, 2023-02-08:
As far as I can see this is now solved after 22ab0e6. Please test again, and reopen this if needed!

closed 2023-02-08 by yorikvanhavre

pablo-mayrgundter, 2023-02-20:
Hi, sorry slow on this!  I didn't realize there was auto-update of the plugin within FreeCAD.. that's nice and easy and will do that next time

I'm seeing more progress now, but still not seeing geometry come thru, with a failure: "<class 'TypeError'>: cannot create 'N4Base19ProgressIndicatorPyE' instances"

Here's the test file:
  https://github.com/bldrs-ai/Share/blob/main/public/index.ifc

Live in bldrs:
  https://bldrs.ai/share/v/p/index.ifc#c:-111.37,14.94,90.63,-43.48,15.73,-4.34

FreeCAD shows the spatial hierarchy, and properties of a selected element, but no geometry and a long stack trace in the reports window.  Similar with haus.ifc and window.ifc in that repo directory.

[image]

Here's my import options:
[image]

And version info:
[image]

berndhahnebach, 2023-02-22:
no problems here ... with file from here https://github.com/aothms/IfcOpenHouse/blob/master/IfcOpenHouse_IFC2x3.ifc

[image]

pablo-mayrgundter, 2023-02-22:
Ah interesting.. you're on FreeCAD 0.21.0.. I'm on 0.20.2, the most recent release from FreeCAD site for OSX.

Do you think that's it?