yorikvanhavre/FreeCAD-NativeIFC #21: Interaction with ifcfile object

berndhahnebach, 2023-02-22:
How is it possible to interact with the IfcOpenShell ifcfile object by python.

Workflow:
- open ifc with this importer
- change label of project by python in the ifcfile object in a way the properties panel gets updated as well

see https://forum.freecad.org/viewtopic.php?t=72345&start=120#p661527

cheers bernd

yorikvanhavre, 2023-02-23:
Copying here the important parts from https://forum.freecad.org/viewtopic.php?p=662877#p662877

1. We offer python methods, ex. ifc_tools.set_property(obj, property, new_value) that would do everything that needs to be done. Similar to the edit_attribute method at [https://github.com/yorikvanhavre/FreeCA ... cts.py#L95](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/ifc_objects.py#L95)

2. We decide that if audacious users decide to modify the ifcopenshell file object directly, they are also responsible to recompute the necessary FreeCAD objects. We should make sure either that recomputing also updates all attributes from the ifcopenshell file (preferred way, I think), or we provide another method for that

berndhahnebach, 2023-02-23:
I can live with 2. if it would work, means if there was a method to just reread similar objects

In the regard of 1. Moult's point seams resonable to me. There are hundreds of possibilities to change IfcOpenShell file object. This ends up in rereading all. 

@Moult what happens on "refresh_ui_data" in BlenderBIM? Does it reread the whole ifcfile object and reset all ui elements?

Moult, 2023-02-23:
In Blender, the UI never reads directly from the IFC. Instead, it reads from a class which in turn reads from the IFC. This class in the middle acts as a cache. Each UI panel has its own cache. The function purges the cache and forces it to refetch data from the IFC.

yorikvanhavre, 2023-02-23:
What I hear from you two is that we need both methods :sweat_smile: 

In FreeCAD we have actually something similar, as the objects have a python object embedded, that does just that, see what needs to be recalculated, etc.

Working on this!

berndhahnebach, 2023-03-01:
What would help a lot would be if it was possible to save a changed ifcfile object to disc in another ifc by some mouse interaction. This way the very annoying path copy to the console and change the backslashes (at work I am on windows) would totaly be away. The save button on right mouse click is only activ if something has been changed in the FreeCAD Property Editor. Furthermore it is only possible to overwrite the ifc and not to save as a new file. 

Besides the cools stuff you have probably in mind this should be quite easy to archieve.

yorikvanhavre, 2023-03-02:
Yes that is easy. I guess we need a "Save as..." option, and after saving the alternate IFC file, ask if that one should become the new FilePath or not. And if not, the project stays marked as modified.

berndhahnebach, 2023-03-02:
AFAIK FreeCAD has "save as" and "save a copy" to make a difference between these both.

commit aa850df4 (2023-03-07): Added 'Save As' context menu option - fixes #21