yorikvanhavre/FreeCAD-NativeIFC #23: import breaks on a IFC which has missing entrys

berndhahnebach, 2023-02-24:
On a huge ifc from a architect I got problems. I have been used this ifc on various viewers including Solibrie, BIMCollabZoom, BIMVision and FreeCAD and BlenderBIM, as well as CAD Allplan. Noone returned big problems and all parts seams correct, BUT

dozens of objects missing the ObjectPlacement. Somehow all the programme did not have a problem with it. As an example some code attached ... 

´´´
Python 3.10.9 | packaged by conda-forge | (main, Feb  2 2023, 20:14:58) [MSC v.1929 64 bit (AMD64)] on win32
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> 
>>> ### Begin command Std_Open
>>> import importIFCparse
>>> importIFCparse.open(u"C:/Users/BHA/Desktop/fehlerhaftes_ifc_laed_nicht_nativ_ifc/4130_ARC_ROHM01_20230203.ifc")
>>> import ifcopenshell
>>> ifcfile = ifc = ifcopenshell.open("C:/Users/BHA/Desktop/fehlerhaftes_ifc_laed_nicht_nativ_ifc/4130_ARC_ROHM01_20230203.ifc")
>>> print(ifc.by_type("IfcProject")[0])
>>> ### End command Std_Open
>>> 
>>> ifcfile[409407]
#409407=IfcSlab('1GQsvNK6n9e9nfjgTH$n8Z',#11,'Beton','Dach, 22 cm Stärke
22 Beton','Dach 22cm',#395870,#409405,$,.ROOF.)
>>> ifcfile[409407].ObjectPlacement
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 136, in __getattr__
    self.wrapped_data.get_argument(idx), self.wrapped_data.file
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\ifcopenshell_wrapper.py", line 4833, in get_argument
    return _ifcopenshell_wrapper.entity_instance_get_argument(self, *args)
RuntimeError: Instance #395870 not found
>>> ifcfile[395870]
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\file.py", line 313, in __getitem__
    return entity_instance(self.wrapped_data.by_id(key), self)
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\ifcopenshell_wrapper.py", line 4515, in by_id
    return _ifcopenshell_wrapper.file_by_id(self, id)
RuntimeError: Instance #395870 not found
>>> 
>>> 
´´´
the native IFC importer has problems with this. It stops creating shapes if it meets such an object. ATM I can not share the file on the internet as it was not created by us but by another company.

berndhahnebach, 2023-02-24:
```
10:25:47  ERROR:root:unhandled exception occurred setting attribute name for #60386=IfcSlab('0uOv8$f1f948AnmH38W1bV',#11,'Beton','Geschossdecke mit Dach, 41 cm Stärke
Beton 35 cm
Dämmung 6 cm','Dach Überstand EG 41-34cm',#60109,#60384,$,.ROOF.)
Traceback (most recent call last):
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 477, in _
    attr_value = self[i]
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 242, in __getitem__
    self.wrapped_data.get_argument(key), self.wrapped_data.file
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\ifcopenshell_wrapper.py", line 4833, in get_argument
    return _ifcopenshell_wrapper.entity_instance_get_argument(self, *args)
RuntimeError: Instance #60109 not found
10:25:47  ERROR:root:unhandled exception occurred setting attribute name for #69084=IfcSlab('3r4oWN7Ir0DvAASUVQBnIK',#11,'Beton','EG Geschossdecke, 36 cm Stärke
Betondecke 30 cm
Dämmung 6 cm','Dach über TG unter Hof 36cm',#68975,#69082,$,.ROOF.)
Traceback (most recent call last):
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 477, in _
    attr_value = self[i]
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 242, in __getitem__
    self.wrapped_data.get_argument(key), self.wrapped_data.file
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\ifcopenshell_wrapper.py", line 4833, in get_argument
    return _ifcopenshell_wrapper.entity_instance_get_argument(self, *args)
RuntimeError: Instance #68975 not found
10:25:47  ERROR:root:unhandled exception occurred setting attribute name for #69333=IfcSlab('1wHWDacgb9bxRJrQAbS9bs',#11,'Beton','EG Geschossdecke, 36 cm Stärke
Betondecke 30 cm
Dämmung 6 cm','Dach über TG unter Hof 36cm',#69116,#69331,$,.ROOF.)
Traceback (most recent call last):
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 477, in _
    attr_value = self[i]
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 242, in __getitem__
    self.wrapped_data.get_argument(key), self.wrapped_data.file
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\ifcopenshell_wrapper.py", line 4833, in get_argument
    return _ifcopenshell_wrapper.entity_instance_get_argument(self, *args)
RuntimeError: Instance #69116 not found
10:25:47  ERROR:root:unhandled exception occurred setting attribute name for #69474=IfcSlab('119pegi$P7kuNtHzwQsR0y',#11,'Beton','EG Geschossdecke, 34 cm Stärke
Betondecke 28 cm
Dämmung 6 cm','Dach über TG unter Hof 34cm',#69365,#69472,$,.ROOF.)
Traceback (most recent call last):
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 477, in _
    attr_value = self[i]
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 242, in __getitem__
    self.wrapped_data.get_argument(key), self.wrapped_data.file
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\ifcopenshell_wrapper.py", line 4833, in get_argument
    return _ifcopenshell_wrapper.entity_instance_get_argument(self, *args)
RuntimeError: Instance #69365 not found
10:25:47  ERROR:root:unhandled exception occurred setting attribute name for #69711=IfcSlab('0tmE2KZ2P97gA4wg_CIA_Q',#11,'Beton','EG Geschossdecke, 34 cm Stärke
Betondecke 28 cm
Dämmung 6 cm','Dach über TG unter Hof 34cm',#69506,#69709,$,.ROOF.)
Traceback (most recent call last):
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 477, in _
    attr_value = self[i]
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 242, in __getitem__
    self.wrapped_data.get_argument(key), self.wrapped_data.file
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\ifcopenshell_wrapper.py", line 4833, in get_argument
    return _ifcopenshell_wrapper.entity_instance_get_argument(self, *args)
RuntimeError: Instance #69506 not found
10:25:47  ERROR:root:unhandled exception occurred setting attribute name for #69948=IfcSlab('2xqt_xqSvA59UrtlO6eLuS',#11,'Beton','EG Geschossdecke, 36 cm Stärke
Betondecke 30 cm
Dämmung 6 cm','Dach über TG unter Hof 36cm',#69743,#69946,$,.ROOF.)
Traceback (most recent call last):
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 477, in _
    attr_value = self[i]
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 242, in __getitem__
    self.wrapped_data.get_argument(key), self.wrapped_data.file
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\ifcopenshell_wrapper.py", line 4833, in get_argument
    return _ifcopenshell_wrapper.entity_instance_get_argument(self, *args)
RuntimeError: Instance #69743 not found
[... cut ...]
10:28:24  DEBUG: ifc_tools.get_coin: Invalid iterator
10:28:24  DEBUG: ifc_tools.get_coin: Invalid iterator
10:28:24  DEBUG: ifc_tools.get_coin: Invalid iterator
10:28:24  DEBUG: ifc_tools.get_coin: Invalid iterator
10:28:28  DEBUG: ifc_tools.get_coin: Invalid iterator
10:28:28  DEBUG: ifc_tools.get_coin: Invalid iterator
10:28:28  DEBUG: ifc_tools.get_coin: Invalid iterator
10:28:28  Recompute failed! Please check report view.
10:28:28  Imported 4130_ARC_ROHM01_20230203.ifc ( 22.86 Mb ) in 02:51
```

berndhahnebach, 2023-02-24:
[image]

berndhahnebach, 2023-02-24:
Attached a IFC which I did edit to get similar behaviour. ... For some reason the tree view does not show the recompute problem mark on this object.

`
10:40:11  ERROR:root:unhandled exception occurred setting attribute name for #80=IfcSlab('2JwE5lcKXFKvWOrg3uyxKD',#5,'Structure','',$,#77,#79,$,.NOTDEFINED.)
Traceback (most recent call last):
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 477, in _
    attr_value = self[i]
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 242, in __getitem__
    self.wrapped_data.get_argument(key), self.wrapped_data.file
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\ifcopenshell_wrapper.py", line 4833, in get_argument
    return _ifcopenshell_wrapper.entity_instance_get_argument(self, *args)
RuntimeError: Instance #77 not found
10:40:11  Imported no_placement_edited.ifc ( 0.0 Mb ) in 00:00
`

[no_placement_edited.ifc.txt](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/files/10822753/no_placement_edited.ifc.txt)

[image]

berndhahnebach, 2023-02-24:
yep fully inside ifcopenshell and I am able to reproduce this in ifcopenshell only.

Have you tried normal FreeCAD importer? No warning nothing from ifcopenshell ... 

Same for the monster big file, it opens in BlenderBIM and BlenderBIM does not complyain. At least not in a way I have noticed it. 

```
11:01:46  Opening 'C:/Users/BHA/Desktop/fehlerhaftes_ifc_laed_nicht_nativ_ifc/no_placement_edited.ifc'... done.
11:01:46  Building types and relationships table...
11:01:46  done.
11:01:46  Parsing 4 BIM objects...
11:01:46  1 / 4 object #80 : IfcSlab no layer found IfcSlab (arch) 2k 1 solids : ID80 Structure 
11:01:46  2 / 4 object #81 : IfcSite (arch) skipped.
11:01:46  3 / 4 object #83 : IfcBuilding (arch) skipped.
11:01:46  4 / 4 object #85 : IfcBuildingStorey (arch) skipped.
11:01:46  Processing Struct relationships...done
11:01:46  Processing Arch relationships...done.
11:01:46  Finished importing 0.0 MB in 2 seconds, or 497 s/MB

```

berndhahnebach, 2023-02-24:
ifcopenshell validate finds it ...
```

>>> ### Begin command Std_Open
>>> import importIFCparse
>>> importIFCparse.open(u"C:/Users/BHA/Desktop/fehlerhaftes_ifc_laed_nicht_nativ_ifc/no_placement_edited.ifc")
>>> ifcfile = ifc = ifcopenshell.open("C:/Users/BHA/Desktop/fehlerhaftes_ifc_laed_nicht_nativ_ifc/no_placement_edited.ifc")
>>> print(ifc.by_type("IfcProject")[0])
>>> Gui.SendMsgToActiveView("ViewFit")
>>> ### End command Std_Open
>>> 
>>> json_logger = ifcopenshell.validate.json_logger()
>>> ifcopenshell.validate.validate(ifcfile, json_logger)
>>> 
>>> json_logger.statements
[{'level': 'error', 'message': 'Invalid attribute value', 'type': 'schema', 'instance': #80=IfcSlab('2JwE5lcKXFKvWOrg3uyxKD',#5,'Structure','',$,#77,#79,$,.NOTDEFINED.), 'attribute': 'IfcSlab.ObjectPlacement'}]
>>> 
>>> 
```

berndhahnebach, 2023-02-24:
open ifc with latest native ifc module, AFIAK this is what happens in importIFC ...

```
import ifcopenshell, os, Part
from ifcopenshell import geom
settings = ifcopenshell.geom.settings()
settings.set(settings.USE_BREP_DATA,True)
settings.set(settings.SEW_SHELLS,True)
settings.set(settings.USE_WORLD_COORDS,True)

p=ifcfile[80]
r = p.Representation
print(p, '\n', r)
cr = ifcopenshell.geom.create_shape(settings, p)
brep = cr.geometry.brep_data
shape = Part.Shape()
shape.importBrepFromString(brep)
Part.show(shape)
```

no error here ... Shape shows in FreeCAD ...

```
Python 3.10.9 | packaged by conda-forge | (main, Feb  2 2023, 20:14:58) [MSC v.1929 64 bit (AMD64)] on win32
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> import ifcopenshell, os, Part
>>> from ifcopenshell import geom
>>> settings = ifcopenshell.geom.settings()
>>> settings.set(settings.USE_BREP_DATA,True)
>>> settings.set(settings.SEW_SHELLS,True)
>>> settings.set(settings.USE_WORLD_COORDS,True)
>>> 
>>> 
>>> p=ifcfile[80]
>>> r = p.Representation
>>> print(p, '\n', r)
#80=IfcSlab('2JwE5lcKXFKvWOrg3uyxKD',#5,'Structure','',$,#77,#79,$,.NOTDEFINED.) 
 #79=IfcProductDefinitionShape($,$,(#78))
>>> cr = ifcopenshell.geom.create_shape(settings, p)
>>> brep = cr.geometry.brep_data
>>> shape = Part.Shape()
>>> shape.importBrepFromString(brep)
>>> Part.show(shape)
<Part::PartFeature>
>>> 
```

yorikvanhavre, 2023-02-24:
This might be the issue: https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/ifc_tools.py#L467

berndhahnebach, 2023-02-25:
> This might be the issue: https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/ifc_tools.py#L467

No it is not it is this line ... https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/e7bd20e6eedf3afd703ad122cc9996a7747b6a82/ifc_tools.py#L199

open ifc and run

`ifcfile[80].get_info().items()`

will result in 

```

15:41:54  ERROR:root:unhandled exception occurred setting attribute name for #80=IfcSlab('2JwE5lcKXFKvWOrg3uyxKD',#5,'Structure','',$,#77,#79,$,.NOTDEFINED.)
Traceback (most recent call last):
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 477, in _
    attr_value = self[i]
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\entity_instance.py", line 242, in __getitem__
    self.wrapped_data.get_argument(key), self.wrapped_data.file
  File "C:\0_BHA_privat\progr\FreeCAD\FreeCAD_0.21.xxxxx\bin\lib\site-packages\ifcopenshell\ifcopenshell_wrapper.py", line 4833, in get_argument
    return _ifcopenshell_wrapper.entity_instance_get_argument(self, *args)
RuntimeError: Instance #77 not found
```

berndhahnebach, 2023-02-25:
that explains why it is not a problem ... the value is needed here because it is one item, but the value will not be used later ...

berndhahnebach, 2023-02-25:
Ahh I missed reading the code. It would be used to create a FreeCAD property. It should be created too even if the referenced entry is missing. It should be mentioned that the entry is missing.

berndhahnebach, 2023-02-26:
IMHO IfcOS should not raise a not handled exception. If this would be handled by IfcOpenShell we could put it in a try: except:  ...

https://github.com/IfcOpenShell/IfcOpenShell/issues/2811

berndhahnebach, 2023-02-26:
this would work with uptodate IfcOpenShell ... 

replace this line by the following code ... https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/e7bd20e6eedf3afd703ad122cc9996a7747b6a82/ifc_tools.py#L199

```
# info_ifcentity = ifcentity.get_info()
# the above raises an unhandled excption on corrupted ifc files
# thus workaround
attribs = []
for anumber in range(20):
    try:
        attr = ifcentity.attribute_name(anumber)
    except Exception:
        break
    # print(attr)
    attribs.append(attr)

info_ifcentity = {}
for attr in attribs:
    try:
        value = getattr(ifcentity, attr)
    except Exception as e:
        value = "Error: {}".format(e)
        # print(e)
    # print(value)
    info_ifcentity[attr] = value

for attr, value in info_ifcentity.items(): 
```

whats your opinion ...

berndhahnebach, 2023-02-26:
this would fix the unhandled IfcOpenShell exception https://github.com/berndhahnebach/FreeCAD-NativeIFC/commit/529d055c44f24a5e3d68eb9c25be24b938fc8e94

We could temporary use this until this gets some fix ... https://github.com/IfcOpenShell/IfcOpenShell/issues/2811

yorikvanhavre, 2023-02-27:
That seems like a good solution. But then I'd put that in a separate function (ex. get_attributes() ) so if this gets fixed we just switch back

berndhahnebach, 2023-02-27:
same opinion here :-)

https://github.com/yorikvanhavre/FreeCAD-NativeIFC/pull/26 see comment I added another two small changes ...

yorikvanhavre, 2023-02-28:
Fixed with #26

closed 2023-02-28 by yorikvanhavre