yorikvanhavre/FreeCAD-NativeIFC #25: Import mod "shapemode" without any representation at all

berndhahnebach, 2023-02-24:
How about another "shapemode" ... no geometry at all. Means the ifc will be parsed only and objects will be created in the regard of strategy pref. This is helpful for very large files if one only needs the ifcopenshell file object or some attributes but no geometry. Or geometry of a certyain element or spatial structure. Thus only the needed geometry could be loaded after import by right click load in tree view.

berndhahnebach, 2023-02-24:
this new ifc module is really starting to make a lot of fun and if I read the road map chrismas seams not far away even it has just pased a few weeks ago ...

yorikvanhavre, 2023-02-24:
No geometry at all is interesting, I never thought about that... But would anybody use it? It seems weird to use FreeCAD for that. But I'd be curious to test the speed anyway ;)

This could have another interesting possibility: 1) you load the file without geometry. 2), in the background, while you are already exploring the file, with the UI responsive, the geometry is gathered and constructed

berndhahnebach, 2023-02-24:
I do use it on every day basis allready. I have some small importer for this which just pares the ifc by ifcopenshell. At work I am on Windows. If I would like to parse a ifc in ifcopenshell I need to switch all backslashes from the copied path. This is cumbersome. So ATM I open the file in freecad and eventually do have a ifcopenshell file object handle in python console. like this https://github.com/yorikvanhavre/FreeCAD-NativeIFC/pull/24 but with empty tree view. 

If this would be possible in this module, I could open the ifc, would have the filehandle in python console and the tree but no geometry. Afterwards I could only load the geometry I need. 

Why Do I need this? We do use BIMTester as QualityGate for all alphnumerical data. If something goes wrong I do need to check properties. For this I do not need geometry in a lot cases.

yorikvanhavre, 2023-02-24:
Yes I see... This would also allow to do interesting things such as:
- open the IFC file
- perform some tests and sanity checks
- only after that, display the geometry...

berndhahnebach, 2023-02-24:
yeah, actually if retrieving the coin representation is commented in the code it is possible ATM, but loading the it later does not work at all.

berndhahnebach, 2023-02-24:
or open the ifc file and just do a validation https://wiki.osarch.org/index.php?title=IfcOpenShell_code_examples#File_validation

yorikvanhavre, 2023-03-09:
Now we have:
- a "no representation" mode in the import dialog
- a "load representation" context menu option for objects with no representation
- loading the representation of an object autoloads the representations of all its children too
So now you can fully load a file without any representation, do all the work you need there, and at any point in time turn the 3D representation on

commit 28013d13 (2023-03-09): Added 'no representation' import mode - fixes #25
closed 2023-03-09 by yorikvanhavre