yorikvanhavre/FreeCAD-NativeIFC #27: handle unreferenced objects

berndhahnebach, 2023-02-28:
ATM a unreferenced object does not get a IfcObject in FreeCAD after import. Best would be if they get a own group right under the IfcProject object.

[unreferenced_box.ifc.txt](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/files/10855201/unreferenced_box.ifc.txt)

yorikvanhavre, 2023-03-02:
Good point... We definitely want to support unreferenced objects. I guess we should start with an `get_unrefereced_objects(ifcfile)` function then we see how and where we render them...

berndhahnebach, 2023-03-02:
do we have this in standard importer?

yorikvanhavre, 2023-03-08:
No because the standard importer just import all of the objects. Here, we have different import strategies so we don't always render all objects.

But when we build the initial structure, we read all the shapes that are included into it. Maybe what we should do:

* get all products: ifcfile.by_type("IfcProduct")
* compare this list with the one we processed already
* if some of them aren't processed, that means they are out of the initial structure. pop a dialog asking the user if they want to import them
* if yes, process them individually and put them in a group

Moult, 2023-03-08:
I'd recommend copying the strategy that BBIM uses to load elements, it's taken quite a bit of refinement over some time to learn what is valuable to load and what isn't.

yorikvanhavre, 2023-03-09:
@Moult you told me once you could choose import strategies in BBIM, if you preferred to lead everything, or only certain items, etc... I cannot find that back when loading an IFC file in Blender. Has it been removed?

Moult, 2023-03-22:
@yorikvanhavre you have to "Enable Advanced Mode" when loading a project.

yorikvanhavre, 2024-03-18:
We can now consider this fixed by be713ee

I simplified a bit the handling of orphans... There is no more "orphans group", orphan objects simply appear under the project. So orphan objects are always loaded by default, but any new object will always be added to the project. In other words, you can load files with orphan objects, but you cannot create orphan objects (they will always be tied to a project).

I don't think it's desirable or advantageous to create orphan objects, or let's say there is no advantage to leave them unrelated to the project. If such necessity arises, though, it wouldn't be much to adapt.