yorikvanhavre/FreeCAD-NativeIFC #28: geom iterator is not valid but ifcopenshell is able to return valid brep geometry

berndhahnebach, 2023-03-01:
On several models from Architects I do have problem to import a geometry into FreeCAD native IFC. They work flawless with FreeCAD standard importer and with BlenderBIM. Attached a very simple file which shows the problem. Open the file with standard FreeCAD importer looks great. Open the file with native importer. The geometry is missing. After import in native importer run the following code. It returns False, thus no geometry ... 

```
import ifcopenshell
import ifc_tools
from ifcopenshell import geom
settings = ifc_tools.get_settings(ifcfile)
iterator = ifcopenshell.geom.iterator(settings, ifcfile, 1, include=[ifcfile[83]])
iterator.initialize()
```

[problem_nativeIFC.ifc.txt](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/files/10858893/problem_nativeIFC.ifc.txt)

I have not posted this on IfcOpenShell ... Just waiting for your opinion Yorik.
#

berndhahnebach, 2023-03-01:
BTW: the ifc is such a BuildinElementPart example. The Wall is a IfcWall without geometry and all wall layer are attached by a IFCRELAGGREGATES. All wall layer objects use IfcBuildingElementPart.

berndhahnebach, 2023-03-02:
I did a first step ... 
- #83 has a representation, the axis as 2D Polyline, thus filter_elements() does not pass the aggregates but the 2D Polyline to get the iterator. This is for sure not what we would like to have. I will have a look. 

The reason why the iterator is invalid for a 2D Polyline is another story ... Bu I am not interested in this ATM.

berndhahnebach, 2023-03-02:
Here we are if the aggregates are used instead the 2D Polyline in filter_elements() it import all walls  :-)

[image]

berndhahnebach, 2023-03-02:
added to #29
https://github.com/yorikvanhavre/FreeCAD-NativeIFC/pull/29/commits/207ef5581064842389a3e0c24855d26d7e907966

Moult, 2023-03-02:
The fix does not look correct. It doesn't take into account contexts, ordering, or other 2D elements. I'm happy to have a code review session together with you @berndhahnebach if you'd like?

berndhahnebach, 2023-03-02:
I am aware this only fixes this case. Since I do no know a all these possible cases I just fixed this. I am happy if you could give us some hints how to better handle such cases. It seams you have been gone through all this before since in BlenderBim mostly all this imports fine.

Since I have family best for me is working hours Zürich time 9:00 to 16:30 or late evening hours Zürich time 20:30 to 22:30.

Moult, 2023-03-02:
https://24timezones.com/difference/zurich/sydney It seems like our timezones don't overlap much at good times (I'd prefer 9am-5pm Sydney time). Is that possible?

Can you help explain where filter_elements and other functions in ifc_tools.py is used? I'm struggling to understand the overall "flow" of logic right now.

berndhahnebach, 2023-03-02:
9am-5pm Sydney time is 11pm(the day before) - 8am in the morning. 

Would be 8 or 8:30 am ok for you as well just for one day ;-) ?  That would be 10pm or 10:30pm for me which would be ok. 

BTW: good morning Dion ... I am just going to bed now ... :-)

berndhahnebach, 2023-03-02:
> https://24timezones.com/difference/zurich/sydney It seems like our timezones don't overlap much at good times (I'd prefer 9am-5pm Sydney time). Is that possible?
>
> Can you help explain where [...]

I will give it a try tomorrow ... But I have just understood it myself today ... started to hack on geometry parsing today by fixing this. It was all done by Yorik.

Moult, 2023-03-02:
Maybe a code review with @yorikvanhavre ? Is the timezone the same or is it possible to do 9-5pm Sydney time?

yorikvanhavre, 2023-03-03:
i'm on the same time zone as Bernd. 8h30 sydney time is fine!

yorikvanhavre, 2023-03-03:
guys i caught covid :/ think i'm going to stay put for a few days... Don't hesitate to do it without me, i'll catch up later!

Moult, 2023-03-06:
@yorikvanhavre no rush, we can wait. Get well soon!

yorikvanhavre, 2023-03-07:
Ok seems like I'm more or less out of it :)
What about tomorrow or thursday?

closed 2023-03-09 by yorikvanhavre