yorikvanhavre/FreeCAD-NativeIFC #30: split ifc_tools in pure ifc tools and tools including FreeCAD objects or methods

berndhahnebach, 2023-03-02:
Would you support this? 

I would like to move all methods which would run on pure IfcOpenShell without FreeCAD obj or FreeCAD methods (except such methods from importIFChelper which also would run on pure IfcOpenShell) in a separate module. These ones could be used outside of FreeCAD without importing FreeCAD. This would make it much easier to reuse thesm in other tools and programme. 

"ifc_tools" would be split in "ifc_tools_ifc" and "ifc_tools_fc" or if you prefere other names they does not really matter for me.

yorikvanhavre, 2023-03-03:
yes, in any case, sooner or later that module would need to be split. we can also make an ifc_tools folder module and inside, different submodules..

Moult, 2023-03-06:
I think it is also worthwhile considering what can be upstreamed to ifcopenshell.util.

yorikvanhavre, 2023-03-07:
excellent idea. We should have a look and upstream everything we can!

yorikvanhavre, 2023-03-22:
I'm looking at this again, and actually almost all of our functions deal with FreeCAD :sweat_smile: 
There are a couple that can be isolated, though. I think it's best to first separate then into a separate file, then see from there what should go upstream...

yorikvanhavre, 2024-03-18:
I will close this one because the code is getting very consolidated and well split now and there is not much point in splitting any further...