yorikvanhavre/FreeCAD-NativeIFC #32: Import error: from ifcopenshell import api

ebrahimraeyat, 2023-03-17:
Hello.
from ifcopenshell import api
<class 'ImportError'>: cannot import name 'api' from 'ifcopenshell' (G:\freecad\git\build\bin\lib\site-packages\ifcopenshell\__init__.py)

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32165 (Git)
Build type: Release
Branch: master
Hash: 0da14beb27357cc61584335bacbf47f3ef0cc66b
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * beso
  * BIM_Workbench 2021.12.0
  * civilTools
  * dodo 1.0.0
  * etabs_api
  * FreeCAD-Help
  * FreeCAD-NativeIFC 2022.10.20
  * git.7z
  * momen
  * OSAFE 2022.5.29
[/code]

thanks

yorikvanhavre, 2023-03-20:
No error here... Where did you get your IfcOpenShell from?
Maybe this is more an error for https://github.com/IfcOpenShell/IfcOpenShell

Moult, 2023-03-21:
Install instructions may be found here https://blenderbim.org/docs-python/ifcopenshell-python/installation.html

yorikvanhavre, 2023-03-22:
Closing, then! :)

closed 2023-03-22 by yorikvanhavre