yorikvanhavre/FreeCAD-NativeIFC #34: Use IfcOpenShell's 2D engine

yorikvanhavre, 2023-03-20:
Use IfcOpenShell to generate SVG code.

yorikvanhavre, 2023-03-21:
@Moult are IfcConvert's svg output capabilities available from Python? I haven't found that anywhere, and I have the impression that BBIM uses its own system to get an SVG view from a mesh, right? 

Looking at https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.7.0/src/blenderbim/blenderbim/bim/module/drawing/svgwriter.py now but the magic doesn't seems to be there...

yorikvanhavre, 2023-03-22:
Thanks! I begin to have a picture.

Yes, doing the annotation work in python is also what we do in FreeCAD. There is so much little detail there to adjust that it makes all sense, and it's still globally relatively fast (much faster than the actual HLR work)

So basically we need to work with a svg serializer:

```python
settings = ifcopenshell.geom.settings(DISABLE_TRIANGULATION=True, STRICT_TOLERANCE=True, INCLUDE_CURVES=True)
buffer = ifcopenshell.geom.serializers.buffer()
serialiser = ifcopenshell.geom.serializers.svg(buffer, settings)
serialiser.setFile(ifcfile)
it = ifcopenshell.geom.iterator(settings, ifcfile, multiprocessing.cpu_count(), include=elements)
if it.initialize():
    while True:
        serialiser.write(it.get())
        if not it.next():
            break
serialiser.finalize()
results = buffer.get_value()
```

Next question then (Thanks for your patience :sweat_smile:) How do you set the projection direction, and/or cut plane?

yorikvanhavre, 2023-03-22:
Ok I think  found the answer myself:

```python
serialiser.addDrawing(
        (x,y,z), # location
        (x,y,z), # z axis (view direction)
        (x,y,z), # x axis
        "Test", # drawing name
        True)    # include projection
```

Moult, 2023-03-22:
Well actually no, the projection direction and cut plane is set in the geometry of the IfcAnnotation object with ObjectType="DRAWING". We established a convention to use a box representation to represent the "section box" of the drawing. So everything is stored in the model and the geometry of the section box is in the model too, so it can be used immediately for intersection tests. It's best to create a drawing in BBIM to see what I mean.

yorikvanhavre, 2023-03-23:
Ok. it's pretty easy to add such an object (all the data can be obtained from an arch section plane). How do you inform the serializer of which drawing it should use? `seraliser.setElevationRefGuid()` ? Is there an different setting if you want a section or just a projection (no cut lines)? Or is that handled automatically?

yorikvanhavre, 2023-03-23:
> Is there an different setting if you want a section or just a projection (no cut lines)? Or is that handled automatically?

Ok found it myself I guess..
```python
serialiser.setAutoElevation(True)
serialiser.setAutoSection(True)
```
I guess we'll need to play a bit with those settings. Probably the best path forward is to use ifcopenshell.draw:

```python
settings = ifcopenshell.draw.draw_settings()
settings.drawing_guid = "XXXX" #  the guid of the drawing to use
svg_result = ifcopenshell.main(settings, [ifcfile])
```