yorikvanhavre/FreeCAD-NativeIFC #35: allow to change the IFC schema

yorikvanhavre, 2023-03-21:
* query available schemas from ifcopenshell
* add a drop-down list to IFC_documents
* add function to change the schema - see https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.7.0/src/blenderbim/blenderbim/bim/module/patch/operator.py

yorikvanhavre, 2023-03-21:
@Moult @aothms is there a way to obtain from Python a list of IFC schemas supported by an instance of IfcOpenShell? Couldn't figure out...

Moult, 2023-03-21:
`ifcopenshell.ifcopenshell_wrapper.schema_names()`

yorikvanhavre, 2023-03-22:
As per https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.7.0/src/ifcpatch/ifcpatch/recipes/Migrate.py#L46-L49 :

```python
file_patched = ifcopenshell.file(schema=newschema)
migrator = ifcopenshell.util.schema.Migrator()
for element in ifcfile:
    migrator.migrate(element, file_patched)
```

commit 7890ada1 (2023-03-28): Finished implementing schema migration - closes #35
closed 2023-03-28 by yorikvanhavre