yorikvanhavre/FreeCAD-NativeIFC #41: [Issue] Sync object visibility between tree-view and 3d-view

balrobs, 2023-04-26:
Steps to reproduce:

1. Import MyHouse.ifc (created with Blenderbim)
2. Select Slab001 in the tree view and press space bar to hide the slab
3. Notice that Slab001 is hidden in the 3d view and is greyed out in the tree view. So far so good!
4. The storey object EG is marked for a recompute (???) Do a recompute.
5. Now select Slab001 in the tree view and press space bar to show the slab
6. The slab turns visible in the 3d view but stays greyed out in the tree view. No recompute visible (???)

[MyHouse.ifc.txt](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/files/11334694/MyHouse.ifc.txt)

yorikvanhavre, 2023-05-04:
The above problem is solved by [8b93a15](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/commit/8b93a15e98a6fe21bdb997e0e92676246c34be26)
However, objects with a Group property still build a coin representation automatically from their children, no matter if using the extension or not. This will be hard to track...
So at the moment the 1x1x1 shape trick at https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/ifc_tools.py#L702 is still used. I'll leave this open until I find a better idea...

yorikvanhavre, 2023-05-16:
Investigating further, even if the Group property is renamed to something else, they still build a coin representation from their children.

yorikvanhavre, 2024-03-07:
This is now solved in FreeCAD main codebranch

closed 2024-03-07 by yorikvanhavre