yorikvanhavre/FreeCAD-NativeIFC #42: [Issue] Wrong result when changing object placement

balrobs, 2023-04-26:
Steps to reproduce:
1. Import MyHouse.ifc (created with Blenderbim) - see issue #41
2. Select Wall001 and translate it along the x axis by changing its position from x/y/z=600cm/0cm/0cm to x/y/z=800cm/0cm/0cm
3. This is the wrong result:
[image]
4. Note that pressing undo gives then the expected result.
[image]

yorikvanhavre, 2023-04-28:
This is solved with https://github.com/yorikvanhavre/FreeCAD-NativeIFC/commit/9dc768756b8ec1e7641d64e2b76df8d8f5ba555d ;)

balrobs, 2023-04-30:
Hi Yorik,
unfortunately I can't see any improvement here with latest NativeIFC-version!
```
OS: Windows 11 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32985 (Git)
Build type: Release
Branch: master
Hash: 5cd377f931312ede305e4c764ba2028b20eaacfd
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * dodo 1.0.0
  * FreeCAD-NativeIFC 2022.10.20
  * FreeCAD_SketchArch
  * Help 1.0.3
```

yorikvanhavre, 2023-05-03:
Oh indeed... Upon doing this the rotation changes from 90 to 180.
There is likely some problem in our matrixes :sweat_smile:

closed 2023-05-04 by yorikvanhavre

balrobs, 2023-05-04:
Problem solved :-)
Thanks Yorik
This Addon is going to become awseome!

commit e4e9d487 (2023-05-04): Fixed rotations - fixes #42