yorikvanhavre/FreeCAD-NativeIFC #43: Error when creating a default project structure

balrobs, 2023-05-04:
Don't know if it's me missing the right workflow or if it's a bug:

1. with latest BIM-addon and NativeIFC-addon create a new empty project
2. activate BIM-addon
3. from the menu select 3D/BIM->Project
4. close dialog <default-sructure> with Yes (should create a default structure: IfcProject->IfcSite->IfcBuilding->IfcBuildingStorey)
5. this is what I see in the tree view:
[image]
6. and these are the errors in the report view:
```
Traceback (most recent call last):
  File "C:\Users\balrobs\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_viewproviders.py", line 72, in claimChildren
    return self.Object.Group
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'Group'
13:24:35  Running the Python command 'BIM_Project' failed:
Traceback (most recent call last):
  File "C:\Users\balrobs\AppData\Roaming\FreeCAD\Mod\BIM\.\BimWrappedTools.py", line 64, in Activated
    project = ifc_tools.create_document(FreeCAD.ActiveDocument)
  File "C:\Users\balrobs\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 90, in create_document
    site = aggregate(Arch.makeSite(), obj)
  File "C:\Users\balrobs\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 918, in aggregate
    product = create_product(obj, parent, ifcfile)
  File "C:\Users\balrobs\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 974, in create_product
    exportIFC.ifcbin = exportIFCHelper.recycler(ifcfile, template=False)

recycler.__init__() got an unexpected keyword argument 'template'
```

yorikvanhavre, 2023-05-04:
You need a more recent version of FreeCAD unfortunately... In [5276114](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/commit/5276114b05c448442851b35e96c2d120b418af3c) I added a better error message.
Unfortunately that's all I could do for now, as this currently uses the Arch IFC exporter. Later on we'll recode that entirely and depend only on IfcOpenShell, that should avoid this kind of problems.

A new release of FreeCAD is coming soon, otherwise you can simply replace your Arch/exportIFCHelper.py with this one: https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Arch/exportIFCHelper.py

yorikvanhavre, 2023-05-16:
fixed with #46

closed 2023-05-16 by yorikvanhavre