yorikvanhavre/FreeCAD-NativeIFC #47: FreeCAD freeze when moving IFC object

balrobs, 2023-05-17:
Hi Yorik!
I found another strange behaviour:
1. Import this IFC-file [IFC_Test1.ifc.txt](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/files/11498222/IFC_Test1.ifc.txt) exportet from RFEM with this options:
[image]
2. Import seems ok.
3. When I select object "Member 1" and launch the transform-command I would expect to see the transform gizmo near "Member 1" and not at the 0,0,0 position.
4. Now I can move the "Member 1", but when I press OK to exit the task FreeCAD freezes.

PS: BlenderBim handles the file without problems

yorikvanhavre, 2023-05-17:
I can't reproduce the problem, for me moving the object using the gizmo works fine. Is there any error message in the output window?

About the gizmo being placed at (0,0,0), that's normal, because that's where the placement of the object is (entity #42). Does it happen differently in BlenderBIM? (I can't test right now, need to recompile it)

balrobs, 2023-05-17:
No, unfortunately no error in the report-view.
But changing the z-value of the Position freezes FreeCAD with errors:
 
[image]

About gizmo problem you are right. Blender shows gizmo at (0,0,0) too.

balrobs, 2023-05-20:
That's really weired ...
in order to narrow down the problem I tried this:
1. update FreeCAD to latest weekly-build (btw I'm on windows) and the addon's Native-IFC and BIM to the latest versions
2. uninstall all other addons
3. Import IFC_Test1.ifc with nativeIFC

Result:
1. when I click on the existing z-value (0,00 cm) of the Position to select it everything is fine
2. when I type 5 instead of 0,00 I see 5cm in the corresponding field and "Member 1" on the screen translates by 5cm. So far everything is ok.
3a. when I then click on the 3d-Viewport or press Enter FreeCAD freezes
3b. when I continue to change the z-value by pressing another number on the keyboard (i.e. 0) I see the value 50cm in the input field and "Member 1" translates to its new position in the 3d-Vieport. At the same time the report view shows the errors I mentioned in my previous message.
4. At this point I can jumb to 3a or 3b

I did another test that maybe can help:
1. import IFC_Test1.ifc
2. save it as IFC_Test1.FCStd (FreeCAD file)
3. close an reopen IFC_Test1.FCStd

Here I see this error:
```
11:30:32  Imported IFC_Test1.ifc ( 0.0 Mb ) in 00:00
11:30:52  Traceback (most recent call last):
  File "C:\Users\Robert\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_objects.py", line 58, in onChanged
    self.edit_placement(obj)
  File "C:\Users\Robert\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_objects.py", line 178, in edit_placement
    result = ifc_tools.set_placement(obj)
  File "C:\Users\Robert\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 866, in set_placement
    scale = ifcopenshell.util.unit.calculate_unit_scale(ifcfile)
  File "C:\Users\Robert\Downloads\FreeCAD_weekly-builds-33236-2023-05-18-conda-Windows-x86_64-py310\bin\lib\site-packages\ifcopenshell\util\unit.py", line 453, in calculate_unit_scale
    units = file.by_type("IfcUnitAssignment")[0]
<class 'AttributeError'>: 'NoneType' object has no attribute 'by_type'
```

balrobs, 2023-05-22:
At first glance I can't notice any difference with the new update.
[image]

balrobs, 2023-05-22:
I dis this other test that maybe can help:
1. load  IFC_Test1.ifc into BlenderBIM
2. in BlenderBIM duplicate "Member 1" to optain "Member 2"
3. save the IFC-file in BB and import it in FC
4. notice that in FC "Member 1" is red and "Member 2" is grey, while in BB they both are red
5. when you change Position to "Member 2" it works as expected
6. when you change Position to "Member 1" FC freezes
Here i append the IFC-file with both Members:
"Member 1" = original and somehow corrupted object
"Member 2" = copied and working object
[IFC_Test1_with_copied_beam.ifc.txt](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/files/11533726/IFC_Test1_with_copied_beam.ifc.txt)

balrobs, 2024-03-15:
Hi Yorik!
I have seen the latest progress in the development of the IFC add-on and I am very happy about it.
So I tried to reproduce this problem with the latest updates (FreeCAD weekly bundle for Windows ver. 2024/03/11 and NaitiveIFC add-on ver. 2023.10.27). The problem seems to have been solved in the meantime, so I think the issue can be closed.
Best regards and thank you for your work!