yorikvanhavre/FreeCAD-NativeIFC #48: Drag & Drop failed - no module named 'ifcopenshell.api.spatial.unassigned_container'

balrobs, 2023-05-22:
When I try to move a newly created Wall001 into the "My Storey" container by drag and drop I run into this error:
[image]
When I drag and release it over the existing wall (I know this doesn't make sense) I run into a "Access violation" <Exeption>

yorikvanhavre, 2023-05-25:
Yeah mine is 0.7.0 too, but it has new commits and fixes almost every day :sweat_smile:

balrobs, 2024-03-17:
Retested with
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * FreeCAD-NativeIFC 2023.10.27
```
and issue is still there:
Steps to reproduce:

1. Create an empty Freecad file
2. File->Import [my_test.ifc.txt](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/files/14627248/my_test.ifc.txt) as nativeIFC with standard import options. (File was createdb with Blenderbim)
3. Import is ok
[image]
4. Try to drag Wall03 into the "My Storey"-Container
5. Freecad hard crashes

yorikvanhavre, 2024-03-18:
No crash for me here:

[image]

I noticed a small bug though, the object is not removed from the Site when dragging it onto the storey, that is fixed with commit 95e87f3

Not sure if that's the origin of your crash, though.. Can you retest?

yorikvanhavre, 2024-03-18:
Great! Thanks for confirming!

BTW that files gives errors when opened in IFC++, and also the objects appear black for me when opened in FreeCAD. But not sure what exactly is happening...

closed 2024-03-18 by yorikvanhavre