yorikvanhavre/FreeCAD-NativeIFC #51: scaling issue of openings?

smr02, 2023-08-10:
I've created a beam from the BlenderBIM demo project with a void and imported it to FreeCAD. 

In the screenshot you can see how the Void/opening shows as a very small cube in FreeCAD. 
Attached the IFC file. 

[image]
[image]

[test_demo.ifc.txt](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/files/12311358/test_demo.ifc.txt)

Moult, 2023-08-10:
A heads up that for these types of openings you should typically not use "openings", you should instead use boolean difference voids.

smr02, 2023-08-10:
@Moult, right. I'm still trying to figure out my workflow. 
I guess, FreeCAD can't display a boolean difference void yet? Can it create them? 

[image]

luzpaz, 2023-08-10:
@smr02 do you mind pasting your [About](https://wiki.freecad.org/About) info ? TIA

smr02, 2023-08-10:
@luzpaz 

```
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git) Snap 727
Build type: Release
Branch: tag: 0.21.0
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD-NativeIFC 2022.10.20
  * BIM 2021.12.0
```

luzpaz, 2023-08-16:
yorikvanhavre is AFK.. If you don't hear a response from a week from today please bump this ticket again, @smr02 TIA

yorikvanhavre, 2023-09-01:
Sorry for the delay!

The very small cube is probably due to a workaround we use right now, that creates such a "default" shape to prevent objects to have a null shape (because of a bug in FreeCAD that auto-populates such shapes). But I'll look further into this one...

About boolean ops, indeed you can see the result but not yet edit them. Coming soon ;)