yorikvanhavre/FreeCAD-NativeIFC #52: Installed Addon not working on windows and Linux

Mannshoch, 2023-10-06:
I found this and tried using `https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/doc/installation.md`

1. uninstalled already installed FreeCAD
1. Installed FreeCAD
   * Windows: `winget install FreeCAD.FreeCAD` (without root)
   No request for installing IfcOpenShell
   * Linux `flatpack install org.freecadweb.FreeCAD`
   Followed by a request to install IfcOpenShell - said YES
1. followed instructions `https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/doc/installation.md`
2. Open an IFC Native
3. Error appear:
   * Windows: `class 'ImportError'>: cannot import name 'api' from 'ifcopenshell' (C:\Users\%username%\AppData\Local\Programs\FreeCAD 0.21\bin\lib\site-packages\ifcopenshell\__init__.py)  `
   * Linux: `<class 'ImportError'>: IfcOpenShell not built for 'linux/64bit/python3.10'`

luzpaz, 2023-10-06:
Full [About](https://wiki.freecad.org/About) info please

Mannshoch, 2023-10-06:
* Linux:
```
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at f670854)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * FreeCAD-NativeIFC 2022.10.20
  * BIM 2021.12.0
```

 * Windows:
```
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * Assembly3 0.12.0
  * BIM 2021.12.0
  * dodo 1.0.0
  * FreeCAD-NativeIFC 2022.10.20
  * sheetmetal 0.2.63
```

yorikvanhavre, 2023-10-09:
> Linux: <class 'ImportError'>: IfcOpenShell not built for 'linux/64bit/python3.10'

Ok that one is easy. Your version of ifcopenshell is not built for the same python version of FreeCAD (3.10). I don't know where your ifcopenshell comes from (is it included in the flatpack? that would be a serious problem). The easiest way is to head here: https://blenderbim.org/docs-python/ifcopenshell-python/installation.html and download a py3.10 version. You might need to find and remove the old one though. It should just be a matter of locating the 'ifcopenshell' folder and deleting it.

> Windows: class 'ImportError'>: cannot import name 'api' from 'ifcopenshell' (C:\Users\%username%\AppData\Local\Programs\FreeCAD 0.21\bin\lib\site-packages\ifcopenshell\__init__.py)

That one is less easy to debug because I don't have a windows install to check, but more than probably a similar reason: ifcopenshell not properly installed. Maybe try and grab a windows version from the page above (beware that your windows version uses python 3.8, not 3.10).

Mannshoch, 2023-10-09:
The Windows one was easy I replaced the mentioned directory with the downloaded from blenderbim. 
On  the other side. I do not know how Flatpak works and where to replace a directory.

yorikvanhavre, 2023-10-10:
We could do some investigating inside the BIM or NativeIFC addon and advise the user better on what to do...

You could try this to know the ifcopenshell location: Inside the FreeCAD python console, type:
```python
import ifcopenshell
print(ifcopenshell.__file__)
```
But maybe you can simply drop the new ifcopenshell folder somewhere FreeCAD will find it, ex. in one of the folders shown by this:
```python
import sys
for p in sys.path:
    print(p)
```

Mannshoch, 2023-10-13:
This command
```
import ifcopenshell
print(ifcopenshell.__file__)
```
create this output on FreeCAD on Flatpack
```
 import ifcopenshell
Traceback (most recent call last):
  File "/app/lib/python3.10/site-packages/ifcopenshell/__init__.py", line 54, in <module>
    from . import ifcopenshell_wrapper
  File "/app/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/app/lib/python3.10/site-packages/ifcopenshell/ifcopenshell_wrapper.py", line 13, in <module>
    from . import _ifcopenshell_wrapper
  File "/app/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ImportError: /app/lib/python3.10/site-packages/ifcopenshell/_ifcopenshell_wrapper.so: undefined symbol: _ZTV13HdfSerializer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/app/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/app/lib/python3.10/site-packages/ifcopenshell/__init__.py", line 62, in <module>
    raise ImportError("IfcOpenShell not built for '%s'" % python_distribution)
ImportError: IfcOpenShell not built for 'linux/64bit/python3.10'
>>> print(ifcopenshell.__file__)# Gui.runCommand('Std_Open',0)

```

I found IfcOpenShell on
`~/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/AdditionalPythonPackages/py310/ifcopenshell`
`/var/lib/flatpak/app/org.freecadweb.FreeCAD/current/active/files/lib/python3.10/site-packages/ifcopenshell`

Moult, 2023-10-13:
At a glance it looks like a Flatpak version. Perhaps a glibc incompatibility? Similar to https://github.com/IfcOpenShell/IfcOpenShell/issues/2706

Mannshoch, 2023-10-13:
That's way above my abilities and skills. I'm not able to follow here. If you advise me I follow. But I'm not a developer. For me it's not anymore that important because I have a working FreeCAD on Windows.

yorikvanhavre, 2023-10-16:
Does the flatpak version use local python libs? Could you check this @Mannshoch?
```python
import sys
for p in sys.path:
    print(p)
```
It could give us a hint to where to put another version of ifcopenshell...

I'm not sure what can be done with the flatpak, @adrianinsaval might have an idea here?

Mannshoch, 2023-10-16:
```
/home/USERNAME/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Mod/BIM/./
/home/USERNAME/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Mod/BIM
/home/USERNAME/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Mod/FreeCAD-NativeIFC/./
/home/USERNAME/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Mod/FreeCAD-NativeIFC
/app/freecad/Mod/Web
/app/freecad/Mod/Tux
/app/freecad/Mod/Test
/app/freecad/Mod/TechDraw
/app/freecad/Mod/Surface
/app/freecad/Mod/Start
/app/freecad/Mod/Spreadsheet
/app/freecad/Mod/Sketcher
/app/freecad/Mod/Show
/app/freecad/Mod/Robot
/app/freecad/Mod/ReverseEngineering
/app/freecad/Mod/Points
/app/freecad/Mod/Plot
/app/freecad/Mod/Path
/app/freecad/Mod/PartDesign
/app/freecad/Mod/Part
/app/freecad/Mod/OpenSCAD
/app/freecad/Mod/MeshPart
/app/freecad/Mod/Mesh
/app/freecad/Mod/Measure
/app/freecad/Mod/Material
/app/freecad/Mod/Inspection
/app/freecad/Mod/Import
/app/freecad/Mod/Idf
/app/freecad/Mod/Fem
/app/freecad/Mod/Draft
/app/freecad/Mod/Arch
/app/freecad/Mod/AddonManager
/app/freecad/Mod
/app/freecad/lib
/app/freecad/Ext

/usr/lib/python310.zip
/usr/lib/python3.10
/usr/lib/python3.10/lib-dynload
/app/lib/python3.10/site-packages
/usr/lib/python3.10/site-packages
/home/USERNAME/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/AdditionalPythonPackages/py310
/home/USERNAME/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/AdditionalPythonPackages
/home/USERNAME/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Macro/
/home/USERNAME/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Macro
/app/freecad/Macro
/app/lib/python3.10/site-packages/ifcopenshell/lib/linux/64bit/python3.10
```

yorikvanhavre, 2023-10-16:
Ok thanks, that's interesting...

So flatpak only considers its own subdirs, and the official system-wide python folders managed by the linux distribution. It discards the user-specific folders.

So basically the only way to supply one's own version of IfcOpenShell is to install it system-wide (basically in /usr/lib/python3.10/site-packages). That's of course annoying because it requires superuser permissions and could interfere with what the distro's package manager does (in the case some day distros start providing IfcOpenShell).

I'm not sure what can be done here, let's wait for Adrian to chime in

adrianinsaval, 2024-03-21:
flatpak should already include ifcopenshell, but if it's installable through pip it would make sense to put it in `/home/USERNAME/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/AdditionalPythonPackages/py310`

adrianinsaval, 2024-03-21:
> But I'm also looking at how to build a better system from within the BIM WB to allow users to install and update ifcopenshell.

How do you envision this would work?

yorikvanhavre, 2024-03-21:
> flatpak should already include ifcopenshell

the problem is that it's evolving very fast and would need to be updated more often by the user.

Hm ifcopenshell is indeed available via pip. It lags a few versions behind: https://pypi.org/project/ifcopenshell/ but it should not be a problem (and users can request it to be updated there). Maybe that's the best option though, since all the complications of pip are now handled by the addon manager...