yorikvanhavre/FreeCAD-NativeIFC #53: Background process for Native IFC?

Mannshoch, 2023-10-09:
I'm not sure what word describes the wish better.  

[So I created a User Journey](https://cryptpad.ccc-ch.ch/pad/#/2/pad/view/4zbaN2Oo1kq2Gy5y03VWOYCQykXtj3PBdknVQyGUZ18/embed/)

yorikvanhavre, 2023-10-10:
The idea and indeed background processing is something that has been in a corner of my mind for some time too.

The main problems here are 

1) the scenarios are very different depending on the use case
2) the gain would be partial
3) OpenCasCade has incomplete multithread support

The main resource hogs are two: The actual creation of shapes and the creation of document objects. The latter is what takes most of the loading time. Putting all the geometry into one single object is pretty fast. So the gain we would have in loading a file in background is little. And if we create many objects, they cause a lot of stuff to insert into the Qt events system and this would create lag sooner or later.

As for the heavyweight boolean operations, up to now I've never been able to make them work in a separate thread unfortunately. Probably because of the complexity of the opencascade/python bindings...

Let's leave this open for more ideas of course, but at the moment I see little way to realize this...

Mannshoch, 2023-10-13:
May a bit OffTopic
Thanks to you, the IFC do not any more need 6h to import into FreeCAD. It's really usable.
I really like it. Now I have only the Problem to export it into STEP for CAD that do not understand IFC. (takes still ages for an export, a 32MB IFC creates a 240MB STEP) 
You seem very skilled. Could I make proposal for a blog entry about inside Converting 3D?

Hows the situation of OpenCasCade how does IfcOpenShell fit into it and what do you assume would be the next evolving beside your NativeIFC? May Native STEP/SAT? 
I work with different CAD Files every day but I do not know anything behind that. I only wonder why an IFC/STEP/etc. viewer need seconds to show a file and a normal import take 6h.

yorikvanhavre, 2023-10-16:
I think you don't need FreeCAD for that. Use ifcopenshell directly ;)
It comes with a "IfcConvert" utility that can convert an IFC file to STEP (and other formats). This is really fast.

> I work with different CAD Files every day but I do not know anything behind that. I only wonder why an IFC/STEP/etc. viewer need seconds to show a file and a normal import take 6h

It depends on the app, but I'd say mostly (if we take FreeCAD as an example) there is heavy translation/transformation needed to convert each IFC object into an element of the host application. Now this does not happen for ex. with STEP because that format is handled directly by FreeCAD's engine (OpenCasCade). Ifcopenshell is also based on opencascade, and swallows large IFC files very fast. With nativeIFC, what we're trying to do is skip the translation process, and give FreeCAD a minimal set of operations to do. This is basically also what BlenderBIM does, and so I'm confident we're tracing a path here that puts us apart (ahead?) from other BIM apps ;)

Mannshoch, 2023-10-16:
I tried IfcOpenShell. But I need to remove not used parts from the IFC. There your tool is way more helpful and way faster at the end. 
10 min from import, modify and export vs. 6h until know.