yorikvanhavre/FreeCAD-NativeIFC #54: AttributeError: 'NoneType' object has no attribute 'addObject'

Mannshoch, 2023-10-16:
I'm playing around with the [sample-test files](https://github.com/buildingSMART/Sample-Test-Files/tree/master/IFC%202x3/Medical-Dental%20Clinic) from buildingSMART.

the Clinic_Plumbing.ifc had this import error:
```
16:18:48  Traceback (most recent call last):
16:18:48    File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_viewproviders.py", line 158, in expandChildren
16:18:48      ifc_tools.create_children(
16:18:48    File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 320, in create_children
16:18:48      result.extend(create_child(obj, child))
16:18:48    File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 288, in create_child
16:18:48      child = create_object(element, doc, ifcfile, mode)
16:18:48    File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 262, in create_object
16:18:48      ifc_layers.add_layers(obj, ifcentity, ifcfile)
16:18:48    File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_layers.py", line 95, in add_layers
16:18:48      lay = get_layer(layer, proj)
16:18:48    File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_layers.py", line 55, in get_layer
16:18:48      group = ifc_tools.get_group(project, "IfcLayersGroup")
16:18:48    File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 1189, in get_group
16:18:48      group = add_object(doc, otype="group", oname=name)
16:18:48    File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 448, in add_object
16:18:48      obj = document.addObject(ftype, oname, proxy, vp, False)
16:18:48  AttributeError: 'NoneType' object has no attribute 'addObject'
```

yorikvanhavre, 2023-10-17:
I don't get that error... But nevertheless I fixed the above problem

closed 2023-10-17 by yorikvanhavre

Mannshoch, 2023-10-27:
May a stupid question, but how could I update this? The add-on manager say Native IFC is version 2022-10-20 (a whole year behind)

yorikvanhavre, 2023-10-27:
Oh I think I need to update https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/package.xml manually... Thanks for the hint!

commit b08a3215 (2023-10-17): Better single doc chooser on import dialog + misc bugfixes - fixes #54