yorikvanhavre/FreeCAD-NativeIFC #55: Error Message on opening an IFC Native

Mannshoch, 2023-10-27:
I updated the Addon to 23.10.27

A test import gives this errors:

(I think I had the same Error also with the old version)

```
 flag
 flag2
 No stream support: No module named 'lark'
 Vorheriger Cache-Prozess wurde unterbrochen, wird neu gestartet...
 Git ist deaktiviert, Git-Makros werden übersprungen
 Fehler beim Verbinden mit dem Wiki, FreeCAD kann die Wiki-Makroliste zu diesem Zeitpunkt nicht abrufen
 Traceback (most recent call last):
File "<string>", line 1, in <module>
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_import.py", line 47, in open
insert(filename, doc.Name)
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_import.py", line 74, in insert
prj_obj = ifc_tools.create_document_object(
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 101, in create_document_object
create_children(obj, ifcfile, recursive=True, assemblies=False)
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 311, in create_children
result.extend(create_child(obj, child))
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 289, in create_child
subresult.extend(create_child(child, building))
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 301, in create_child
create_children(
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 311, in create_children
result.extend(create_child(obj, child))
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 301, in create_child
create_children(
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 311, in create_children
result.extend(create_child(obj, child))
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 279, in create_child
child = create_object(element, doc, ifcfile, mode)
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_tools.py", line 253, in create_object
ifc_layers.add_layers(obj, ifcentity, ifcfile)
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_layers.py", line 98, in add_layers
<class 'AttributeError'>: 'NoneType' object has no attribute 'Group'
```

```
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods:
* Assembly3 0.12.0
* BIM 2021.12.0
* dodo 1.0.0
* FreeCAD-NativeIFC 2023.10.27
* sheetmetal 0.3.1

```

yorikvanhavre, 2023-10-27:
I just did a quick fix, but I'm in the middle of more changes, so I'll wait a bit before bumping up the version number...

closed 2023-10-27 by yorikvanhavre

Mannshoch, 2023-10-27:
Please take your time 😁

Sorry to bother you again. But I didn't realize that Freecad was still working. ( I started it only for testing)  50min later I got this error message and the import finalized.

```
15:54:39 DEBUG: No Shape returned for element 122, IfcProject, 11014
15:54:39 Traceback (most recent call last):
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_objects.py", line 129, in execute
ifc_generator.generate_geometry(obj, cached=cached)
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_generator.py", line 73, in generate_geometry
print_debug(obj)
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_generator.py", line 424, in print_debug
if not element.is_a("IfcContext") and not element.is_a(
<class 'AttributeError'>: 'NoneType' object has no attribute 'is_a'
15:54:39 Recompute failed!
15:54:39 Traceback (most recent call last):
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_objects.py", line 129, in execute
ifc_generator.generate_geometry(obj, cached=cached)
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_generator.py", line 73, in generate_geometry
print_debug(obj)
File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_generator.py", line 424, in print_debug
if not element.is_a("IfcContext") and not element.is_a(
<class 'AttributeError'>: 'NoneType' object has no attribute 'is_a'
15:54:39 Recompute failed!
```

yorikvanhavre, 2023-11-10:
This should be fixed with [04abbc9](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/commit/04abbc97480fc394af625669bbec36031115d72c)

Mannshoch, 2023-11-13:
I updated the Addon and reimported the same IFC. 
```
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * BIM 2021.12.0
  * FreeCAD-NativeIFC 2023.10.27
```

```
13:06:34  DirectWrite: CreateFontFaceFromHDC() failed (Gibt einen Fehler in einer Eingabedatei an, zum Beispiel eine Schriftartdatei.) for QFontDef(Family="8514oem", pointsize=10, pixelsize=20, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("8514oem", lfWidth=0, lfHeight=-20) dpi=192
13:06:43  No stream support: No module named 'lark'
13:07:35  DirectWrite: CreateFontFaceFromHDC() failed (Gibt einen Fehler in einer Eingabedatei an, zum Beispiel eine Schriftartdatei.) for QFontDef(Family="8514oem", pointsize=10, pixelsize=20, styleHint=5, weight=75, stretch=100, hintingPreference=0) LOGFONT("8514oem", lfWidth=0, lfHeight=-20) dpi=192
13:11:00  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_import.py", line 48, in open
    insert(filename, doc.Name)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_import.py", line 86, in insert
    ifc_psets.load_psets(prj_obj)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_psets.py", line 232, in load_psets
    load_psets(child)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_psets.py", line 232, in load_psets
    load_psets(child)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_psets.py", line 232, in load_psets
    load_psets(child)
  [Previous line repeated 1 more time]
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_psets.py", line 226, in load_psets
    show_psets(obj)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_psets.py", line 94, in show_psets
    ptype, value = pvalue.split("(", 1)
<class 'ValueError'>: not enough values to unpack (expected 2, got 1)
```

The Import works fast and without problems. After Import, the Screen is empty an it had to Refresh. This refresh took double the time of the import.

yorikvanhavre, 2023-11-16:
The above commit should fix that...

commit b1fab6e5 (2023-10-27): Fixed #55