yorikvanhavre/FreeCAD-NativeIFC #56: <class 'ValueError'>: too many values to unpack (expected 2)

Mannshoch, 2023-11-23:
```
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * FreeCAD-NativeIFC 2023.10.27
```
I uninstalled FreeCAD with User settings, an this FreeCAD runs a portable version in Windows
Is a different IFC file than the last times I wrote Issues.
```
10:31:43  DirectWrite: CreateFontFaceFromHDC() failed (Gibt einen Fehler in einer Eingabedatei an, zum Beispiel eine Schriftartdatei.) for QFontDef(Family="8514oem", pointsize=10, pixelsize=20, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("8514oem", lfWidth=0, lfHeight=-20) dpi=192
10:32:13  DirectWrite: CreateFontFaceFromHDC() failed (Gibt einen Fehler in einer Eingabedatei an, zum Beispiel eine Schriftartdatei.) for QFontDef(Family="8514oem", pointsize=10, pixelsize=20, styleHint=5, weight=75, stretch=100, hintingPreference=0) LOGFONT("8514oem", lfWidth=0, lfHeight=-20) dpi=192
10:33:27  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_import.py", line 48, in open
    insert(filename, doc.Name)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_import.py", line 86, in insert
    ifc_psets.load_psets(prj_obj)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_psets.py", line 232, in load_psets
    load_psets(child)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_psets.py", line 232, in load_psets
    load_psets(child)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_psets.py", line 232, in load_psets
    load_psets(child)
  [Previous line repeated 1 more time]
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_psets.py", line 226, in load_psets
    show_psets(obj)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_psets.py", line 94, in show_psets
    ptype, value = pvalue.split("(", -1)
<class 'ValueError'>: too many values to unpack (expected 2)
```
follow Error flooded the log window
```
10:34:11  Traceback (most recent call last):
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_objects.py", line 129, in execute
    ifc_generator.generate_geometry(obj, cached=cached)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_generator.py", line 65, in generate_geometry
    elements = get_decomposition(obj)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_generator.py", line 316, in get_decomposition
    elements = get_decomposed_elements(element, obj)
  File "C:\Users\%USERNAME%\AppData\Roaming\FreeCAD\Mod\FreeCAD-NativeIFC\.\ifc_generator.py", line 347, in get_decomposed_elements
    for child in ifcopenshell.util.element.get_decomposition(
<class 'TypeError'>: get_decomposition() got an unexpected keyword argument 'is_recursive'
```
```
10:35:09  Recompute failed!
```

yorikvanhavre, 2023-11-23:
I think this indicates a wrong version of IfcOpenShell... Is it recent enough?

yorikvanhavre, 2024-03-18:
Can you test if the problem still appears using a development version of FreeCAD from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds ?

Mannshoch, 2024-03-22:
I'm not anymore able to recreate the same environment. I have a new computer and meanwhile no access to this ifc files anymore.

I currently have freecad installed on Ubuntu from snapcraft beta (0.22-g93439591) from freecad-org and stable (2024-03-22-gda98001c) from realthunder on both a pip install ifcopenshell is requested by the Add-on and also works if I do that 
(running `freecad.pip install ifcopenshell` ).

I performed tests from a file I downloaded from https://github.com/RWTH-E3D/DigitalHub/blob/master/Version_2/DigitalHub_FM-HZG_v2.ifc

With 5GB RAM both FreeCAD Crash without any error message while importing.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36458 (Git) Snap 958
Build type: Release
Branch: main
Hash: 934395913a6f9298f7b56bfe3b58c163f7011186
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Germany (de_DE)
Installed mods: 
  * dodo 1.0.1
  * FreeCAD-NativeIFC 2023.10.27
  * Nodes 0.1.36
  * ose-piping
  * InventorLoader 1.4.0
  * Render 2024.1.23
  * sheetmetal 0.4.6
```

Mannshoch, 2024-04-04:
FreeCad.log 

﻿```

Msg: FreeCAD 0.22.0, Libs: 0.22.0devR36683 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.

Log: Time = 2024-Apr-04 16:00:58
Log: AboutImage = freecadabout
Log: AppDataSkipVendor = true
Log: AppHomePath = /snap/freecad/972/usr/
Log: AppIcon = freecad
Log: AppTempPath = /home/m/snap/freecad/common/temp/
Log: BOOST_VERSION = 1_74
Log: BinPath = /snap/freecad/972/usr/bin/
Log: BuildRepositoryURL = git://github.com/FreeCAD/FreeCAD.git main
Log: BuildRevision = 36683 (Git)
Log: BuildRevisionBranch = main
Log: BuildRevisionDate = 2024/03/30 16:46:25
Log: BuildRevisionHash = 6fbd8916ab4ae045fdaa4e16a3e44d7393b13e80
Log: BuildVersionMajor = 0
Log: BuildVersionMinor = 22
Log: BuildVersionPoint = 0
Log: BuildVersionSuffix = dev
Log: Console = 0
Log: CopyrightInfo = © Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.

Log: Debug = 0
Log: DocPath = /snap/freecad/972/usr/doc/
Log: EIGEN_VERSION = 3.4.0
Log: ExeName = FreeCAD
Log: ExeVendor = FreeCAD
Log: ExeVersion = 0.22.0
Log: LoggingConsole = 1
Log: LoggingFile = 1
Log: LoggingFileName = /home/m/snap/freecad/common/FreeCAD.log
Log: MaintainerUrl = https://www.freecad.org/wiki/Main_Page
Log: OCC_VERSION = 7.7.1
Log: PATH = /snap/freecad/972/usr/sbin:/snap/freecad/972/usr/bin:/snap/freecad/972/sbin:/snap/freecad/972/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/freecad/972/kf5/usr/bin
Log: PYSIDE_VERSION = 5.15.11
Log: PYTHONPATH = 
Log: PYTHON_VERSION = 3.10.12
Log: PythonSearchPath = /snap/freecad/972/usr/lib/python310.zip:/snap/freecad/972/usr/lib/python3.10:/snap/freecad/972/usr/lib/python3.10/lib-dynload
Log: QT_VERSION = 5.15.10
Log: RunMode = Gui
Log: SMESH_VERSION = 7.7.1.0
Log: SplashAlignment = Bottom|Left
Log: SplashInfoColor = #8aadf4
Log: SplashInfoPosition = 6,75
Log: SplashScreen = freecadsplash
Log: SplashTextColor = #8aadf4
Log: StartWorkbench = StartWorkbench
Log: SystemParameter = /home/m/snap/freecad/common/system.cfg
Log: UserAppData = /home/m/snap/freecad/common/
Log: UserCachePath = /home/m/snap/freecad/common/temp/
Log: UserConfigPath = /home/m/snap/freecad/common/
Log: UserHomePath = /home/m/snap/freecad/common
Log: UserMacroPath = /home/m/snap/freecad/common/
Log: UserParameter = /home/m/snap/freecad/common/user.cfg
Log: Verbose = 
Log: XERCESC_VERSION = 3.2.3
Log: Create Application
Log: Run App init script
Log: Init: starting App::FreeCADInit.py
Log: Init:   Searching for modules...
Log: Init:      Initializing /snap/freecad/972/usr/Mod/AddonManager... done
[... 172 lines cut ...]

```

CLI

```

m@VMWareTest:~$ freecad -l
FreeCAD 0.22.0, Libs: 0.22.0devR36683 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.

Adding snap-specific PYTHONPATH to sys.path: /home/m/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/972/lib/python3.10/site-packages:/snap/freecad/972/usr/lib/python3/dist-packages
Sheet Metal workbench loaded
Qt: Session management error: Could not open network socket
flag
flag2
connect failed: No such file or directory
No stream support: No module named 'lark'
Terminated

```

![grafik](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/assets/7795891/3342ce8b-9de2-4786-811d-9ef3d9bfa667)