yorikvanhavre/FreeCAD-NativeIFC #57: Enhancement Request - move Dialog in Task Area, add Stats window and create a Native IFC feature

Mannshoch, 2023-11-23:
In my opinion the Popup Dialog that appear if you open the IFC should be inside the Task Area. 
At the End of the Importing I would Like to see a little statistic about containing data.

I propose to add a Feature to the "Tree View" that represent the original IFC-Filename.
Please add this feature into the context menu:

- Show Stats and Properties
   Count of  Mesh bodies:
   Count of geometric Bodies:
   Some Properties like GPS-Coordinates
   Name of Project
- Hide/show mesh bodies
- Hide/Show geometric Bodies
- Update IFC data from linked file
- Change Linked file path

yorikvanhavre, 2023-11-27:
I'm not sure about the task dialog because the idea of the task dialog is that you can continue to work in the 3D view while it's open. When importing a file, that's not what we want. Having the task panel open with import settings while you can still do stuff in the 3D view is counter-intuitive and could even cause problems and mistakes.

Ideally, import options should be part of the open dialog, like in Blender. But ATM FreeCAD uses the system's open dialog, and there is no easy way to extend it...

I don't understand well what would be the necessity of splitting the file and the project, as one file can contain one and only one project?

Mannshoch, 2023-11-27:
I agree, with the Idea that Importing Options should be inside the FileOpen dialog. 

But I do not understand what you mean with the "while doing stuff" In my opinion the Task panel contain the current work I do and the moment I Click OK the Work is calculated. Normal use of Taskpanel in imho and pretty similar to the popup. 
File -> Import -> select file -> Task bar -> Setup import settings -> OK

After click OK you have two possibilities, like now to freeze whole FreeCAD until the file is imported. Or like #53, Create the Project into the TreeView - like in the mock-up picture above - and add a huge I'm-calculating-spinner. 
May you could also add a [dummy cube](https://s2.qwant.com/thumbr/0x380/c/9/a141b9ded6d7fa8945d015518cf4fb01a1efb1ff7cd7878fdda37faabb8a2f/giphy.gif?u=https%3A%2F%2Fmedia.giphy.com%2Fmedia%2FpT7V1p6umUyn6%2Fgiphy.gif&q=0&b=1&p=0&a=1) to the View to indicate a loading geometry which get replaced the moment the import is finished.

> I don't understand well what would be the necessity of splitting the file and the project, as one file can contain one and only one project?

My Problem is more, that every IFC I Import is Named "Default Site" or "Project" or "Default Project" May you could check if a Project Name exist and If not simply replace it with the FileName.

yorikvanhavre, 2023-12-19:
> My Problem is more, that every IFC I Import is Named "Default Site" or "Project" or "Default Project" May you could check if a Project Name exist and If not simply replace it with the FileName.

That's a good idea. I just added it in the above commit

Moult, 2024-01-09:
A little note that one reason to split project and site is that it represents that the site is aggregated to the project. There may also be other things declared to the project (e.g. type asset libraries) or project libraries.

My personal $0.02 is that IFC data shouldn't be "masked". If it comes in as "Default Site" then that means that's what the data in the IFC contains - and so be it, best to complain upstream to the IFC provider to do a better job with their data.

Mannshoch, 2024-01-09:
> and so be it, best to complain upstream to the IFC provider to do a better job with their data.
Good joke 🤪

Ever tried to tell a client to do something on their IFC where you are responsible for?

I'm – even with much discussion – not able to get a client to reduce IFCs to only the data I need. At first, I do not need FreeCAD because I work with IFC. I need STEP, SAT, IGES, DWG but my clients send me not usable data. Now I have to pick the Data I need my self from IFC.
Most clients I have, does not have the competence to work with the IFCs they use. 

**One problem they have, my clients are customers them self and do not have influence either.**

IFC is coming more and more, but the knowledge seems a bigger problem.

yorikvanhavre, 2024-01-11:
> Ever tried to tell a client to do something on their IFC where you are responsible for?

If you receive an IFC file from, let's say, an architect, with no site defined in it, then yes definitely your architect did a poor job. And yes I totally get that's what architects do and you can't just go there and complain all the time :sweat_smile: 

Also, one IFC file could contain more than one site. I could imagine many scenarios where that could be the case. So it makes full sense to always keep them distinct.

The only junction we can IMHO do, is "mix" the project and the file. Because IFC is really about the contents. It starts with the project. The file itself is just a way to save that contents, it can be just an attribute of the project.