yorikvanhavre/FreeCAD-NativeIFC #59: Add a Part or PartDesign object to a IFC

smr02, 2024-02-04:
I'm trying to add a Part or PartDesign object to a IFC. 

The following code adds a wall to my project: 
```
wall = Arch.makeWall(None,220,400,20)
ifc_tools.aggregate(wall, project)
```

However, when i run  `ifc_tools.aggregate(obj, project)`, where `obj` is a PartDesign object, i get the following error: 

```
>>> ifc_tools.aggregate(obj, project)
Exception ignored in: <function entity_instance.__del__ at 0x7fd06c18bac0>
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAUhQwpr/usr/lib/python3.10/site-packages/ifcopenshell/entity_instance.py", line 132, in __del__
    self.wrapped_data.file = None
  File "/tmp/.mount_FreeCAUhQwpr/usr/lib/python3.10/site-packages/ifcopenshell/entity_instance.py", line 143, in __getattr__
    attr_cat = self.wrapped_data.get_attribute_category(name)
  File "/tmp/.mount_FreeCAUhQwpr/usr/lib/python3.10/site-packages/ifcopenshell/entity_instance.py", line 143, in __getattr__
    attr_cat = self.wrapped_data.get_attribute_category(name)
  File "/tmp/.mount_FreeCAUhQwpr/usr/lib/python3.10/site-packages/ifcopenshell/entity_instance.py", line 143, in __getattr__
    attr_cat = self.wrapped_data.get_attribute_category(name)
  [Previous line repeated 995 more times]
  File "/tmp/.mount_FreeCAUhQwpr/usr/lib/python3.10/site-packages/ifcopenshell/entity_instance.py", line 142, in __getattr__
    INVALID, FORWARD, INVERSE = range(3)
RecursionError: maximum recursion depth exceeded in comparison
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/smr/.local/share/FreeCAD/Mod/FreeCAD-NativeIFC/./ifc_tools.py", line 893, in aggregate
    product = create_product(obj, parent, ifcfile)
  File "/home/smr/.local/share/FreeCAD/Mod/FreeCAD-NativeIFC/./ifc_tools.py", line 944, in create_product
    product = api_run("root.create_entity", ifcfile, ifc_class=ifcclass, name=name)
  File "/home/smr/.local/share/FreeCAD/Mod/FreeCAD-NativeIFC/./ifc_tools.py", line 229, in api_run
    result = ifcopenshell.api.run(*args, **kwargs)
  File "/tmp/.mount_FreeCAUhQwpr/usr/lib/python3.10/site-packages/ifcopenshell/api/__init__.py", line 66, in run
    result = usecase_class(ifc_file, **settings).execute()
  File "/tmp/.mount_FreeCAUhQwpr/usr/lib/python3.10/site-packages/ifcopenshell/api/root/create_entity.py", line 74, in execute
    element = self.file.create_entity(
  File "/tmp/.mount_FreeCAUhQwpr/usr/lib/python3.10/site-packages/ifcopenshell/file.py", line 318, in create_entity
    e = entity_instance((self.schema_identifier, type), self)
  File "/tmp/.mount_FreeCAUhQwpr/usr/lib/python3.10/site-packages/ifcopenshell/entity_instance.py", line 119, in __init__
    e = ifcopenshell_wrapper.new_IfcBaseClass(*e)
  File "/tmp/.mount_FreeCAUhQwpr/usr/lib/python3.10/site-packages/ifcopenshell/ifcopenshell_wrapper.py", line 4374, in new_IfcBaseClass
    return _ifcopenshell_wrapper.new_IfcBaseClass(schema_identifier, name)
ValueError: invalid null reference in method 'new_IfcBaseClass', argument 2 of type 'std::string const &'
```

### Version
```
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35803 (Git) AppImage
Build type: Release
Branch: main
Hash: 90c90f705bec20faf4b8bb419574134c33c7d4b5
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * DynamicData 2.60.0
  * FreeCAD-NativeIFC 2023.10.27
  * BIM 2021.12.0
```

yorikvanhavre, 2024-02-13:
This will be fixed by https://github.com/FreeCAD/FreeCAD/pull/12352

smr02, 2024-03-19:
Unfortunately I can't run this build. (Segmentation fault, same as here: https://github.com/FreeCAD/FreeCAD-Bundle/issues/217). 
I will try again, when there is a new build.

smr02, 2024-03-20:
I've tested it. It's working now!

closed 2024-03-20 by smr02

yorikvanhavre, 2024-03-20:
Thanks for the feedback!