yorikvanhavre/FreeCAD-NativeIFC #62: Support IFC4.3.2 for infrastructure?

RickBrice, 2024-02-17:
Any plans to support IFC4.3.2 for infrastructure? The reason I ask is that I've been adding geometry mapping to IfcOpenShell on the v0.8.0 branch, c++ code, for alignment and linear referencing. 

Ultimately I want to make BlenderBIM or FreeCAD support infrastructure, bridges in particular. I'm thinking something along the lines of road, bridge, and rail workbenches supporting native IFC.

yorikvanhavre, 2024-02-19:
Basically the schemas supported by the NativeIFC addon are what IfcOpenShell supports. So if 4.3.2 is supported by IfcOpenShell, it becomes automatically available here. In fact, I think 4.3.2 is supported already...

Can you give me an example of entity present in that release? We can try...

RickBrice, 2024-02-19:
IfcOpenShell does support 4.3.2 in so much as it generates the parsing code from the express definition of the schema. I am working on the geometry mapping. An example is an IfcAlignment with a representation as IfcCompositeCurve, or subtype. IfcCompositeCurve is a collection of IfcCurveSegment and each curve segment has a parent curve such as IfcLine, IfcCircle, or IfcClothoid. IfcOpenShell does not generate the geometry of the 2D or 3D wire for visualization. IfcOpenShell also does not position entities with IfcLinearPlacement.

yorikvanhavre, 2024-03-12:
I think the best path here is to wait for proper support in IfcOpenShell... Did you signal this issue there?

RickBrice, 2024-03-12:
I'm working on adding support for alignment in IfcOpenShell. The majority of it is complete. Still have some work to do on IfcSegmentReferenceCurve and cant.

Moult, 2024-04-16:
@yorikvanhavre FYI I plan to start testing the v0.8.0 branch after the next release of BBIM.

RickBrice, 2024-04-16:
While there has been significant progress on the v0.8.0 branch with respect to alignments, there are some serious issues with railway cant. If anyone is knowledgeable, or can point me to someone who is, that would be greatly appreciated.