yorikvanhavre/FreeCAD-NativeIFC #64: [Issue] Document containing an IfcProject and FreeCAD objects

balrobs, 2024-03-22:
Steps to reproduce:

1. Start FreeCAD
2. File->Open  [IfcProject_01.ifc.txt](https://github.com/yorikvanhavre/FreeCAD-NativeIFC/files/14725563/IfcProject_01.ifc.txt)  (with Native (ifc_import))
3. Select BIM workbench
4. Notice Lock-Status is ON
5. Save file as IfcProject_01.FCStd
6. Notice Lock-Status is OFF
7. Draw line with line command
8. When I click for the line end point I see this in the Report view
[image]
9. And when I click on the Close-Button to terminate the line command I see this in the Report View
[image]
10. Then FreeCAD crashes

yorikvanhavre, 2024-03-25:
There are two problems here:

1) the lock button doesn't stay on after reloading
2) creating a line in locked mode fails

yorikvanhavre, 2024-03-25:
Problem 1 should be solved by the above commit.

For problem 2, 2D entities are not working well yet, but at least it should be better handled.

balrobs, 2024-03-25:
After an update hee on my site problem 1 unfortunately stil remains. I am follwing the mentioned steps.